fuzzware-fuzzer/fuzzware #4: fuzzware genconfig --fuzz-for

MrMatch246, 2022-06-20:
using the fuzz-for option on genconfig does very strange things 
it somehow results in lots of nonexistant sed accesses when fuzzing afterwards without any manual configurations
(tried it on "examples/pw-recovery/ARCH_PRO/basic_exercises.bin")
is this already implemented?

Scepticz, 2022-06-23:
If you like, you can give implementing this feature a shot and then create a pull request. :-)

Scepticz, 2022-06-24:
I changed the order in which checks are performed so that the fuzzware genconfig `--fuzz-for` argument is being non-zero is detected earlier. The output should be less confusing now.

You may still create a pull request implementing the actual funcitonality, though!

closed 2022-06-24 by Scepticz