fuzzware-fuzzer/fuzzware #7: Running in AFL++ mode still uses AFL

CounterCycle, 2022-09-21:
Hello,

During my testing I noticed that the fuzzer output files didn't all correspond to the AFL++ format even when the --aflpp flag is used. Looking at the pipeline code, the AFLplusplus folder can never be selected. The execution of the fuzzer process in this function https://github.com/fuzzware-fuzzer/fuzzware-pipeline/blob/d868ab800d0e1186ad088699cc081c68219c5401/fuzzware_pipeline/run_fuzzer.py#L45 depends on the value of `AFL_FUZZ`. When this value is defined, it does not consider the usage of AFLplusplus
https://github.com/fuzzware-fuzzer/fuzzware-pipeline/blob/d868ab800d0e1186ad088699cc081c68219c5401/fuzzware_pipeline/run_fuzzer.py#L11-L12

The string ../../emulator/afl/afl-fuzz is always appended onto the end of the directory for the fuzzer. This results in the execution of `path/to/fuzzware/emulator/AFLplusplus/../../emulator/afl/afl-fuzz` rather than `path/to/fuzzware/emulator/AFLplusplus/afl-fuzz`

I'm unsure if any changes need to be made to AFLplusplus to function correctly with fuzzware, or whether simply changing the directory is enough to fix it.

Scepticz, 2022-09-21:
Hello CounterCycle,

you are correct. Thank you for pointing this out. The use plain AFL, so I did not realize that AFL++ is never used.

I am currently on this (it turns out some surrounding changes are required in the pipeline code).

Tobi

Scepticz, 2022-09-21:
I changed the invocation to now actually invoke AFL++ in the pipeline: https://github.com/fuzzware-fuzzer/fuzzware-pipeline/commit/0ded31a9aa4185be2063ddb90836219520de091b

I also fixed a fork server setup issue with AFL++ in the emulator which lead to input instability here: https://github.com/fuzzware-fuzzer/fuzzware-emulator/commit/a70fd8cebf18671a44deeb72474434fa8d746aaf

Let me know in case you encounter any additional issues surrounding the use of AFL++

Scepticz, 2022-09-21:
I am closing the issue with the fixes in place. Feel free to create another one for a different issue

closed 2022-09-21 by Scepticz