fuzzware-fuzzer/fuzzware #16: errors generated when execute ./build_docker.sh or ./install_local.sh

123huahua, 2023-03-06:
The following errors will be prompted when using, whether using docker installation or local installation. One netizen pointed out that it was because of the version of unicorn, but I used the unicorn given in the fuzzware project. Do I need to install the correct version of unicorn?

native_hooks.c:192:34: error: use of undeclared identifier 'UC_HOOK_BLOCK_UNCONDITIONAL'
    res |= uc_hook_add(uc, &tmp, UC_HOOK_BLOCK_UNCONDITIONAL, hook_block_debug, NULL, 1, 0);
                                 ^
native_hooks.c:472:65: error: use of undeclared identifier 'UC_HOOK_BLOCK_UNCONDITIONAL'
    return uc_hook_add(uc, &hook_block_cond_py_handlers_handle, UC_HOOK_BLOCK_UNCONDITIONAL, hook_block_cond_py_handlers, user_data, 1, 0);
                                                                ^
native_hooks.c:823:5: error: implicit declaration of function 'uc_fuzzer_init_cov' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
    uc_fuzzer_init_cov(uc, bitmap, MAP_SIZE);
    ^
native_hooks.c:1124:9: error: implicit declaration of function 'uc_fuzzer_reset_cov' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        uc_fuzzer_reset_cov(uc, 1);
        ^
4 errors generated.
make: *** [native_hooks.o] Error 1
<builtin>: recipe for target 'native_hooks.o' failed
make: Leaving directory '/home/user/fuzzware/emulator/harness/fuzzware_harness/native'
The command '/bin/sh -c make -C $FUZZWARE/emulator/harness/fuzzware_harness/native clean all' returned a non-zero code: 2

Scepticz, 2023-03-07:
Hi 123huahua,

this indeed looks like the previous issue that you commented on: https://github.com/fuzzware-fuzzer/fuzzware-emulator/issues/1#issuecomment-1455296403

Are you sure you have a completely clean docker build? You could try either deleting some related pre-built docker images / layers, or add the `--no-cache` flag to `docker build`.

The underlying issue is that during build, the patched versions of the Unicorn header files are not used. This is mostly the case if Unicorn has been re-installed in a new version. An alternative root cause could be that the Unicorn build/install did not work within the docker container on your system.

My recommendation here would be the same as for the other ticket:
1. Create a clean checkout of the code
2. Delete docker images / add `--no-cache` to docker build
3. Use ./build_docker.sh

Best
Tobi

123huahua, 2023-03-09:
Dear Tobi,

Thank you for your timely reply. Your enthusiasm surprised me. I re-created a virtual machine and re-executed the installation process. This time, there were no previous errors. The previous error may be a problem in the process of installing dependencies and configuring the environment.

Best
123huahua

closed 2023-07-11 by Scepticz