fuzzware-fuzzer/fuzzware #18: Why extended Unicorn Engine with implementation NVIC and SysTick?

LittleNewton, 2023-03-24:
Fuzzware's code is excellent and clean! I have read the Usenix Paper. In chapter 5, the author said the following:

> to achieve precisely reproducible timing behavior, we measure elapsed time by the number of emulated basic blocks. **We also extended the Unicorn Engine with an implementation of the interrupt controller (NVIC) and the system tick timer (SysTick), which are defined in the Cortex-M standard.**

I am very new to Unicorn Engine. Is that mean the original Unicorn Engine did not implement NVIC and SysTick? Or just for the purpose of the work's reproducibility, you choose to reimplement the two components.

Scepticz, 2023-03-24:
Hi LittleNewton,

you are right in your assumption. While the Unicorn does contain some source code fragments around interrupt handling, these are not complete and are not exposed outside. This also makes sense from the Unicorn Engine perspective, as Unicorn Engine does not attempt to emulate device/peripheral behavior outside instruction set architecture (ISA) emulation. This is why the NVIC/SysTick logic was needed.

Tobi

Scepticz, 2023-07-11:
I am closing this issue

closed 2023-07-11 by Scepticz