fuzzware-fuzzer/fuzzware #19: Fuzzing input not consumed for 150000 basic blocks

liyansong2018, 2023-04-14:
Hi, thank you for your excellent work on fuzzware!  When I tried to test my firmware using fuzzware, I encountered the following error

```shell
$ fuzzware pipeline --run-for 24:00:00
INFO:emulator:No function hooks found. Registering no native basic block hook for that
DEBUG:emulator:Calling init_nvic with vtor=0x08000000, num_vecs: 256
DEBUG:emulator:No non-native unconditional basic block hooks registered, not adding global hook
INFO:emulator:Passing control to native code to start emulation. Running for input file '/home/user/fuzzware/targets/my/fuzzware-project/.empty_input'
[FORKSERVER SETUP] It looks like we are not running under AFL, going for single input
Fuzzing input not consumed for 150000 basic blocks, exiting
[ERROR] Emulation stopped using just the prefix input (0: OK (UC_ERR_OK))
[ERROR] Could not retrieve the number of required ticks during discovery forking
[04-14 09:10:07 ERROR] pipeline.py - Failed to perform emulator dry run: Emulator status code != 0
```

Thanks, :)

Scepticz, 2023-04-14:
Hi liyansong2018,

this means that the firmware made no MMIO accesses within the first amount of execution. This normally has one of two reasons:

1. The firmware has a very long boot sequence. In this case, you may need to increase the respective limit (see README_config.yml)
2. The MMIO regions are not configured correctly, so that the MMIO accesses are not registered as MMIO accesses, but rather as normal RAM by the emulator

Tobi

liyansong2018, 2023-04-14:
Sorry, I overlooked that my firmware was divided into two bin files, which means that I did not simulate boot.bin before simulating object.bin. 

Thank you for your reply!

closed 2023-04-17 by liyansong2018