fuzzware-fuzzer/fuzzware #25: can't compile afl successfully

SaerTrial, 2023-07-12:
The reported errors while implementing "./build_docker.sh"
` 
=> ERROR [17/27] RUN UNICORN_QEMU_FLAGS="--python=/usr/bin/python3" | make -C /home/user/fuzzware/emulator/afl clean all                                    0.2s
------                                                                                                                                                            
 > [17/27] RUN UNICORN_QEMU_FLAGS="--python=/usr/bin/python3" | make -C /home/user/fuzzware/emulator/afl clean all:                                               
0.119 make: Entering directory '/home/user/fuzzware/emulator/afl'                                                                                                 
0.127 rm -f afl-gcc afl-fuzz afl-showmap afl-tmin afl-gotcpu afl-analyze afl-as as afl-g++ afl-clang afl-clang++ *.o *~ a.out core core.[1-9][0-9]* *.stackdump test .test test-instr test-instr-argv .test-instr0 .test-instr1 qemu_mode/qemu-2.10.0.tar.bz2 afl-qemu-trace                                                        
0.129 rm -rf out_dir qemu_mode/qemu-2.10.0 .tmp_outputs .tmp_inputs                                                                                               
0.130 make -C llvm_mode clean
0.131 make[1]: Entering directory '/home/user/fuzzware/emulator/afl/llvm_mode'
0.132 rm -f *.o *.so *~ a.out core core.[1-9][0-9]* test-instr .test-instr0 .test-instr1 
0.133 rm -f ../afl-clang-fast ../afl-llvm-pass.so ../afl-llvm-rt.o ../afl-llvm-rt-32.o ../afl-llvm-rt-64.o ../afl-clang-fast++
0.134 make[1]: Leaving directory '/home/user/fuzzware/emulator/afl/llvm_mode'
0.134 make -C libdislocator clean
0.137 make[1]: Entering directory '/home/user/fuzzware/emulator/afl/libdislocator'
0.137 rm -f *.o *.so *~ a.out core core.[1-9][0-9]*
0.139 rm -f libdislocator.so
0.139 make[1]: Leaving directory '/home/user/fuzzware/emulator/afl/libdislocator'
0.139 make -C libtokencap clean
0.141 make[1]: Entering directory '/home/user/fuzzware/emulator/afl/libtokencap'
0.141 rm -f *.o *.so *~ a.out core core.[1-9][0-9]*
0.142 rm -f libtokencap.so
0.143 make[1]: Leaving directory '/home/user/fuzzware/emulator/afl/libtokencap'
0.144 [*] Checking for the ability to compile x86 code...
0.203 /tmp/cczhFTk6.s: Assembler messages:
0.203 /tmp/cczhFTk6.s:10: Error: unknown mnemonic `xorb' -- `xorb %al,%al'
0.204 
0.204 Oops, looks like your compiler can't generate x86 code.
0.204 
0.204 Don't panic! You can use the LLVM or QEMU mode, but see docs/INSTALL first.
0.204 (To ignore this error, set AFL_NO_X86=1 and try again.)
0.204 
0.204 make: *** [test_x86] Error 1
0.204 Makefile:52: recipe for target 'test_x86' failed
0.205 make: Leaving directory '/home/user/fuzzware/emulator/afl'
------
dockerfile:32
--------------------
  30 |     user user
  31 |     run ./get_afl.sh
  32 | >>> run UNICORN_QEMU_FLAGS="--python=/usr/bin/python3" | make -C $FUZZWARE/emulator/afl clean all
  33 |     run make -C $FUZZWARE/emulator/AFLplusplus clean all
  34 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c UNICORN_QEMU_FLAGS=\"--python=/usr/bin/python3\" | make -C $FUZZWARE/emulator/afl clean all" did not complete successfully: exit code: 2`

It seems like that the latest version of clang is not correct or not compatible with the specified afl?

SWW13, 2023-07-13:
It's a docker build, so it should be stable. What operating system and architecture are you using? (`uname -a`)

SaerTrial, 2023-07-13:
Yes, i thought it should be stable as well and OS-agnostic . My setting with the aforementioned problem is `Darwin MacBook-Air.local 22.3.0 Darwin Kernel Version 22.3.0: Mon Jan 30 20:38:43 PST 2023; arm64` while another one that works well is "Linux ubuntu-2204; x86_64 GNU/Linux". 

Furthermore, the docker image can't execute properly even I directly pulled the fuzzware docker image, e.g., observer.start() will emit an OS error ([Errno 38] Function not implemented). In the meanwhile, using my Macbook, I tested the building in a pure ubuntu VM virtualized by UTM, and this issue still persists.

SWW13, 2023-07-13:
Fuzzware is currently only tested on `x86_64` and it looks like there are some issues with `arm64`. The pre-built docker image only supports `x86_64`.

Support for `arm64` needs some investigation of the problems and hopefully only some small fixes.

closed 2023-08-25 by SaerTrial