fuzzware-fuzzer/fuzzware #27: check_afl_requirements() error

zhangpwxwk, 2023-08-16:
Hello:
   I'v build a docker image for fuzzware. When I use fuzzware in docker, there is an error, messages as follow:

zhangpeng@zhangpeng-virtual-machine:~/fuzzware$ ./run_docker.sh 
[*] defaulting to targets_dir '/home/zhangpeng/fuzzware/examples'
[*] defaulting to cmd         '/bin/bash'
[+] Mapping local dir '/home/zhangpeng/fuzzware/examples' into container
[+] Executing command: '/bin/bash'
[+] Running with -it
user@825f07b43f9d:~/fuzzware/targets$ fuzzware pipeline P2IM/CNC/ --skip-afl-cpufreq --run-for 1:00:00
[08-16 08:20:27 INFO] __init__.py - Executing pipeline at 2023-08-16 08:20:27.343517
[08-16 08:20:27 INFO] __init__.py - Got projdir: /home/user/fuzzware/targets/P2IM/CNC
[08-16 08:20:27 INFO] __init__.py - Found 'base_inputs' dir in target directory, using that as base input
[08-16 08:20:27 INFO] __init__.py - Performing initial tests
[08-16 08:20:27 ERROR] __init__.py - Failed: core_pattern check
[08-16 08:20:27 ERROR] __init__.py - 
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Did you configure your system so that a plain fuzzer will run correctly?
For afl, this would be (on the host, as root):
echo core >/proc/sys/kernel/core_pattern
cd /sys/devices/system/cpu
echo performance | tee cpu*/cpufreq/scaling_governor

Try the fuzzing command line from the output above
XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX

Then I read the source code in __init__.py ,I found the error is generated by function:check_afl_requirements() . 

def check_afl_requirements():
    check_failed = False
    try:
        with open("/proc/sys/kernel/core_pattern", "rb") as f:
            contents = f.read(1)
            if len(contents) == 1 and contents == b'|':
                logger.error("Failed: core_pattern check")
                check_failed = True
    except FileNotFoundError:
        pass

    if os.getenv("AFL_SKIP_CPUFREQ") is None:
        try:
            with open("/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor", "rb") as f:
                contents = f.read()
                if not contents.startswith(b'perf'):
                    with open("/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq", "rb") as f:
                        contents = f.read()
                        min_freq = int(contents)
                    with open("/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq", "rb") as f:
                        contents = f.read()
                        max_freq = int(contents)
                    if min_freq != max_freq:
                        logger.error("Failed: min vs max scaling")
                        check_failed = True
        except FileNotFoundError:
            pass

    if check_failed:
        logger.error("\nXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX\nDid you configure your system so that a plain fuzzer will run correctly?\nFor afl, this would be (on the host, as root):\necho core >/proc/sys/kernel/core_pattern\ncd /sys/devices/system/cpu\necho performance | tee cpu*/cpufreq/scaling_governor\n\nTry the fuzzing command line from the output above\nXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX\n")
        exit(1)

Has anyone encountered this problem?
                                                                                                           Thank you

Scepticz, 2023-08-16:
Hi zhangpwxwk,

have you run the suggested command?

```
sudo su
echo core >/proc/sys/kernel/core_pattern
```

Tobi

zhangpwxwk, 2023-08-16:
The result:

***@***.***:~/fuzzware/targets$ echo core >/proc/sys/kernel/core_pattern

bash: /proc/sys/kernel/core_pattern: Read-only file system

***@***.***:~/fuzzware/targets$ sudo su

***@***.***:/home/user/fuzzware/targets# echo core >/proc/sys/kernel/core_pattern

bash: /proc/sys/kernel/core_pattern: Read-only file system

***@***.***:/home/user/fuzzware/targets#

Scepticz, 2023-08-16:
The read-only filesystem usually stems from running the command from within the docker container. The message suggests running the command on the host: `For afl, this would be (on the host, as root):`

Have you tried that as well?

zhangpwxwk, 2023-08-16:
After run the commond on the host as root , fuzzware works, Thank you very much.

***@***.***:~/fuzzware$ echo core >/proc/sys/kernel/core_pattern

bash: /proc/sys/kernel/core_pattern: 权限不够

***@***.***:~/fuzzware$ sudo su

[sudo] zhangpeng 的密码： 

***@***.***:/home/zhangpeng/fuzzware# echo core >/proc/sys/kernel/core_pattern

***@***.***:/home/zhangpeng/fuzzware# su

***@***.***:/home/zhangpeng/fuzzware# su zhangpeng

Even though, it works,I want to know why  we have to  run the command.

closed 2023-08-16 by Scepticz