fuzzware-fuzzer/fuzzware #28: How does fuzzer perform mutation based on the access model?

3kywa1ke2, 2023-08-23:
Hi,

I have a question about fuzzer's mutation. My understanding is that the access model will guide the fuzzer's mutation, such that the fuzzer only mutates the input bytes specified in the access model. Please let me know if this is correct and if so, can you pinpoint the code where fuzzer consume the access model and perform mutation with its guidance?

Thank you in advance!

Scepticz, 2023-08-23:
Hi 3kywa1ke2,

it is the second option. The input bytes are mutated in the vanilla AFL/AFL++ way.

Best
Tobi

3kywa1ke2, 2023-08-24:
Hi Tobi,

Thank you for your reply!

So essentially, the access model has the info about how many bytes are accessed and the fuzzer will fuzz those bytes as a consecutive sequence of bytes. is this statement correct?

Thank you again for your time!

Scepticz, 2023-08-24:
Hi there,

Yes, that statement is correct. You may state it that way in your paper submission if you like. ;-)

Best
Tobi

closed 2023-09-06 by Scepticz