fuzzware-fuzzer/fuzzware #30: Emulate

zhangpwxwk, 2023-09-03:
Hello:
I would like to ask several simple questions. When I want to emulate my own firmware, I need to prepare xxx.bin config.yml and valid_basic_blocks.txt . 
1.Is xxx.elf needed? 
2. How to generate config.yml and valid_basic_blocks.txt files, especially when the source code is not avaliable. Do you any good methods and suggestions?

Thank

Scepticz, 2023-09-03:
1. The ELF file is not needed. Only the .bin file. However, an ELF makes it easier to understand the execution of course, as it is easier to reverse engineer and as you may have access to symbols.
2. `config.yml` can be generated via `fuzzware genconfig` or manually. If you generate it manually, you will need to know where the flash is based and which memory regions are mapped. Note that if you use `fuzzware genconfig` you should still look at the output and sanity check it, as a correct config is not always guaranteed.
3. `valid_basic_blocks.txt` is not really required. It is important mostly for coverage comparisons between different fuzzers, which is especially important in the academic context.

Best
Tobi

zhangpwxwk, 2023-09-17:
Hello Tobi：
     When I emulate my  firmware, some errors occor.
    
    The errors :
    
    should I change the limit of fuzz_consumption_timeout ?
    Do you have any good Ideas？