fuzzware-fuzzer/fuzzware #33: UC_ERR_INSN_INVALID

Lawliar, 2023-09-29:
Hi folks,

There seems to be a problem with decoding the instruction, I tried to run Unicorn with `--debug` flag and the output is as follows:
`
Calling function: Reset_Handler
Basic Block: addr= 0x000000000800c000 (lr=0x0)
Calling function: Reset_Handler
Basic Block: addr= 0x000000000800c000 (lr=0x0)
Basic Block: addr= 0x000000000800c002 (lr=0x0)
        >>> Read: addr= 0x000000000800c0d0 size=4 data=0x20001000 (pc 0x0800c002)
Basic Block: addr= 0x000000000800c008 (lr=0x0)
        >>> Write: addr= 0x00000000e000ef34 size=4 data=0xc0000000 (pc 0x0800c018)
        >>> Write: addr= 0x00000000e000ed88 size=4 data=0x00f00000 (pc 0x0800c032)
[ERROR] Emulation stopped using just the prefix input (10: Invalid instruction (UC_ERR_INSN_INVALID))
[ERROR] Could not retrieve the number of required ticks during discovery forking
`
Is there any way to know which instruction the unicorn failed to decode?

Many thanks in advance!

Scepticz, 2023-09-30:
Hi Lawliar,

you can run any input in `fuzzware emu` using the `-M` flag. This will show you at least the last basic block which Unicorn tried to execute. If I had to guess, if you are running an ARM Cortex-M, it would be something floating pointer number or ARM-v8 related. In case it is floating point and you have a change to recompile the target, then you could use the `-mfloat-abi=soft` flag.

Best
Tobi

Lawliar, 2023-09-30:
Hi Scepticz,

Thanks for you reply! Yeah, I did tried to recompile with the soft fpu, but it looks like ChibiOS, which the firmware has hard coded fpu in assembly as part of the startup code, which causes this problem. there seems no way to remove this. 
Do you think is there anyway to get around this?

Best,
Lawliar

Scepticz, 2023-09-30:
Other than patching it out, not for the moment :-(

Lawliar, 2023-10-09:
Yeah, that is unfortunate. Thanks for the information tho!

closed 2023-10-09 by Lawliar