fuzzware-fuzzer/fuzzware #37: fuzzware cov for finding the highest-coverage input

B03901108, 2023-11-23:
Hi, I wonder if it is possible to find the input of the highest block coverage in a fuzzware-project using the utility `fuzzware cov` (or `fuzzware genstats coverage`).

It seems that `fuzzware cov` can list the inputs that cover a certain symbol without rerunning them. I am curious if this utility can either (1) get the coverage of each input in fuzzware-project/main<the-latest>/.../queue/, (2) list the inputs that cover more than a specified number of blocks, or (3) find the input of the highest block coverage.

Rerunning the inputs would be a last resort. Thank you.

B03901108, 2023-11-23:
I found that `fuzzware cov` runs on the bblset-traces in fuzzware-project/mainX/.../traces/, so I could achieve the above goals by processing those traces as well. However, some inputs generated by secondary fuzzers in .../queue/ have no bblset-traces in .../traces/. Is this intended?

Scepticz, 2023-11-24:
Hi there,

if traces are missing for inputs, then they had either not been generated, yet, or their generation job was swallowed along the way. If it was swallowed, then this would be unintended (could be due to missed inotify triggers?). You can, however, generate these missing traces via `fuzzware gentraces`.

Tobi

closed 2023-12-19 by B03901108