fuzzware-fuzzer/fuzzware #42: Problem with the gdbserver feature

hoodiney, 2024-04-26:
Hi!

I'm currently trying to play with the gdbserver feature offered by Fuzzware. Here are the steps I took:

1. I first run `fuzzware pipeline examples/P2IM/Console` to fuzz test the Console target.
2. Then I try to replay the input found during fuzzing, with gdbserver port assigned and one breakpoint. 
`fuzzware emu -c examples/P2IM/Console/fuzzware-project/main001/config.yml -d -v -g 3333 -b 0x00002cc2 examples/P2IM/Console/fuzzware-project/main001/fuzzers/fuzzer1/queue/id:000298,src:000000+000094,op:splice,rep:64,+cov `
3. Finally I use a `gdb-multiarch` client to attach to the gdbserver.

The issue I encountered was when I tried to use the "si" or "ni" instructions, the client would hang, and be unable to continue. I'm wondering if it is because I used the feature in the wrong way, it would be great to hear from the authors before I mess up the codebase.

Scepticz, 2024-04-26:
Hi hoodiney,

the GDB server is a very experimental feature that we added, but is not well tested. So there may be issues that you are running into (and maybe have to fix on the way). If you want to hit less of these hurdles, I would suggest the python-based debugger version. However, digging more into that feature and fixing any issues / making it work well and then submitting a pull request would of course be appreciated as well!

Tobi

hoodiney, 2024-04-27:
Hi Tobi,

Thank you very much for the timely reply! I'm looking into the details of the gdbserver implementation now, hopefully I could find a way to fix the issues, it would be of great help for debugging some complex crashes. 

Best.