fuzzware-fuzzer/fuzzware #45: Current version fails to run pw-recovery examples: Exception: 'modeling' worker spawned by WorkerPool is not running anymore

cyseen, 2024-11-13:
Hey,

running the pw-recovery examples of the current version of Fuzzware in Docker results in the following error:

user@832bcc797e12:~/fuzzware/targets/pw-recovery/ARCH_PRO$ fuzzware pipeline --skip-afl-cpufreq .
[11-13 19:28:08 INFO] __init__.py - Executing pipeline at 2024-11-13 19:28:08.390331
[11-13 19:28:08 INFO] __init__.py - Got projdir: /home/user/fuzzware/targets/pw-recovery/ARCH_PRO
[11-13 19:28:08 INFO] __init__.py - Could not find 'base_inputs' directory in target base dir, defaulting to generic initial inputs
[11-13 19:28:08 INFO] __init__.py - Performing initial tests
[11-13 19:28:08 INFO] __init__.py - Initial tests successful
[11-13 19:28:08 WARNING] pipeline.py - Found an existing project directory under /home/user/fuzzware/targets/pw-recovery/ARCH_PRO/fuzzware-project. Moving it now
[11-13 19:28:08 WARNING] pipeline.py - Removing old backed up project: /home/user/fuzzware/targets/pw-recovery/ARCH_PRO/fuzzware-project_old
[11-13 19:28:08 INFO] pipeline.py - logging pipeline output to: /home/user/fuzzware/targets/pw-recovery/ARCH_PRO/fuzzware-project/logs/pipeline.log
[11-13 19:28:08 INFO] pipeline.py - Dry-running emulator to check functionality..

[11-13 19:28:08 INFO] run_target.py - Running target with
config path: /home/user/fuzzware/targets/pw-recovery/ARCH_PRO/config.yml
input path: /home/user/fuzzware/targets/pw-recovery/ARCH_PRO/fuzzware-project/.empty_input
Extra args: ['-v']

add_mmio_region called! hooking 0x40000000 - 0x60000000
[FORKSERVER SETUP] It looks like we are not running under AFL, going for single input
[DISCOVERY FORK PARENT] Got number of ticks to step: 694
Ran into instruction limit of 692 at 0x000011ca - exiting
[+] Initial constant execution (including optional prefix input) done, starting input execution.
Running without a fork server

>>> Ran out of fuzz

Exited without crash at 0x00001210 - If no other reason, we ran into one of the limits
[11-13 19:28:09 INFO] pipeline.py - Emulator dry-run successful!
[11-13 19:28:09 INFO] pipeline.py - Adding pipeline level observers
[11-13 19:28:09 INFO] pipeline.py - Initializing worker pool
[11-13 19:28:09 INFO] pool.py - [WorkerPool] Starting redis server
[11-13 19:28:09 WARNING] pool.py - Redis ping returned: True
[11-13 19:28:09 INFO] pool.py - Redis server successfully started on port 58113
[11-13 19:28:09 WARNING] pool.py - Redis ping returned: True
[11-13 19:28:09 INFO] pool.py - [WorkerPool] Spawning additional modeling worker
[11-13 19:28:09 INFO] pool.py - [WorkerPool] enabling logging for modeling worker!
[11-13 19:28:09 INFO] pool.py - [WorkerPool] Spawning additional gen worker
[11-13 19:28:09 INFO] pool.py - [WorkerPool] enabling logging for gen worker!
Traceback (most recent call last):
  File "/usr/local/bin/fuzzware", line 33, in <module>
    sys.exit(load_entry_point('fuzzware-pipeline', 'console_scripts', 'fuzzware')())
  File "/home/user/fuzzware/pipeline/fuzzware_pipeline/__init__.py", line 1070, in main
    args.func(args, leftover)
  File "/home/user/fuzzware/pipeline/fuzzware_pipeline/__init__.py", line 273, in do_pipeline
    pipeline = Pipeline(args.target_dir, args.project_name, args.base_inputs, args.num_local_fuzzer_instances, args.disable_modeling, write_worker_logs=not args.silent_workers, do_full_tracing=args.full_traces, config_name=args.runtime_config_name, timeout_seconds=timeout_seconds, use_aflpp=args.aflpp)
  File "/home/user/fuzzware/pipeline/fuzzware_pipeline/pipeline.py", line 265, in __init__
    self.worker_pool = WorkerPool(self, write_logs=write_worker_logs)
  File "/home/user/fuzzware/pipeline/fuzzware_pipeline/workers/pool.py", line 77, in __init__
    self.check_running_procs()
  File "/home/user/fuzzware/pipeline/fuzzware_pipeline/workers/pool.py", line 84, in check_running_procs
    raise Exception("'{}' worker spawned by WorkerPool is not running anymore".format(queue_name))
Exception: 'modeling' worker spawned by WorkerPool is not running anymore

worker_modeling_00.log:
 
user@832bcc797e12:~/fuzzware/targets/pw-recovery/ARCH_PRO/fuzzware-project/logs$ cat worker_modeling_00.log 
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/user/.virtualenvs/fuzzware-modeling/lib/python3.10/site-packages/fuzzware_modeling/rq_worker.py", line 7, in <module>
    import angr
  File "/home/user/.virtualenvs/fuzzware-modeling/lib/python3.10/site-packages/angr/__init__.py", line 46, in <module>
    from .sim_procedure import SimProcedure
  File "/home/user/.virtualenvs/fuzzware-modeling/lib/python3.10/site-packages/angr/sim_procedure.py", line 8, in <module>
    from cle import SymbolType
  File "/home/user/.virtualenvs/fuzzware-modeling/lib/python3.10/site-packages/cle/__init__.py", line 11, in <module>
    from .backends import (
  File "/home/user/.virtualenvs/fuzzware-modeling/lib/python3.10/site-packages/cle/backends/__init__.py", line 1, in <module>
    from .backend import ALL_BACKENDS, Backend, ExceptionHandling, FunctionHint, FunctionHintSource, register_backend
  File "/home/user/.virtualenvs/fuzzware-modeling/lib/python3.10/site-packages/cle/backends/backend.py", line 8, in <module>
    import archinfo
  File "/home/user/.virtualenvs/fuzzware-modeling/lib/python3.10/site-packages/archinfo/__init__.py", line 26, in <module>
    from .arch_aarch64 import ArchAArch64
  File "/home/user/.virtualenvs/fuzzware-modeling/lib/python3.10/site-packages/archinfo/arch_aarch64.py", line 20, in <module>
    class ArchAArch64(Arch):
  File "/home/user/.virtualenvs/fuzzware-modeling/lib/python3.10/site-packages/archinfo/arch_aarch64.py", line 46, in ArchAArch64
    cs_arch = _capstone.CS_ARCH_ARM64
AttributeError: module 'capstone' has no attribute 'CS_ARCH_ARM64'. Did you mean: 'CS_ARCH_ARM'?

Thank you!

chunlin007, 2024-11-14:
This issue was introduced by a new commit (https://github.com/fuzzware-fuzzer/fuzzware/commit/b0e8d31b9e56d885e4e8130a49f348f1d701e780). The author tried to update the version of ubuntu and angr in dockerfile(ubuntu18.04 to ubuntu22.04 / angr-8.19.10.30 to angr-9.2.61), but some features used in new version of angr are different from old one (some changes in capstone can refer to https://github.com/capstone-engine/capstone/blob/next/docs/cs_v6_release_guide.md) , causing error during pipeline.  Here are some solutions:
(1) Go back to commit 4463fad72593649f983a6895589f7a5210ffce09, just ignore the update of dockerfile;ls fuzz
(2) Make some changes to archinfo package: 
In fuzzware docker:
`source /home/user/.virtualenvs/fuzzware-modeling/bin/activate
`
open file "vim /home/user/.virtualenvs/fuzzware-modeling/lib/python3.10/site-packages/archinfo/arch_aarch64.py"
and change "cs_arch = _capstone.CS_ARCH_ARM64" to "cs_arch = _capstone.CS_ARCH_AARCH64"   (line 46)
then open file "/home/user/.virtualenvs/fuzzware-modeling/lib/python3.10/site-packages/archinfo/arch_s390x.py" 
change "cs_arch = _capstone.CS_ARCH_SYSZ" to "cs_arch = _capstone.CS_ARCH_SYSTEMZ" 

Of course, choosing the right version of capstone may be the best way.

Scepticz, 2024-11-19:
Hi cyseen and chunklin007,

I pinned capstone to version `5.0.3` in commit https://github.com/fuzzware-fuzzer/fuzzware/commit/28ce2dc3f888ecf1bf8d8da314a88b755d0ecc88

This should resolve the compatibility issues with angr. Let me know if you encounter issues here

Tobi

chunlin007, 2024-11-19:
Perfect! After changing the version of capstone to 5.0.3, fuzzware can run smoothly.

closed 2024-11-20 by Scepticz